ros/urdf_tutorial #13: Jade Release

wjwwood, 2015-04-23:
This is included in desktop-full: http://wiki.ros.org/jade/Planning/Maintenance

@DLu do you have time to do this release?

DLu, 2015-04-23:
I just started working on it today. I'm waiting to hear back from @isucan about his maintainer status.

isucan, 2015-04-24:
David, i am unable to do anything for this until next week. If you'd like
to become the maintainer for this, that would be very helpful. Doing this
release alone would also be very helpful. Thanks!

DLu, 2015-04-24:
Hrm. So urdf_tutorial has a runtime dependency on pr2_description because it uses [meshes from the pr2 gripper](http://wiki.ros.org/urdf/Tutorials/Building%20a%20Visual%20Robot%20Model%20with%20URDF%20from%20Scratch#Finishing_the_Model)

Should I 
A) Save local versions of the meshes
B) Wait for a [jade release](https://github.com/PR2/pr2_common/issues/239) of pr2_description? 
C) Remove the run_depends but not the code. 
D) Panic. 
E) Something else
?

wjwwood, 2015-04-24:
It would be nice not to have to depend on `pr2_description`, but it seems like `pr2_common` could be released if `convex_decomposition` (https://github.com/ros/convex_decomposition/issues/2) and `ivcon` (https://github.com/ros/ivcon/issues/2) get released. If it can get done in the short term, I'd say `B)`.

DLu, 2015-04-29:
Not sure if there's going to be any movement on the jade release of pr2_common. 

If I went with A, would the best method be a literal copy and paste of the files, or something else, do you think?

wjwwood, 2015-04-29:
Copying it would be fine. Though, I think part of the tutorial is to show that you can use resources from outside of the package you are in. Given the list of not so good options, I guess we go with A.

wjwwood, 2015-04-29:
Hold on that, @TheDash gave me permission to get pr2_description released. I'll do it now.

wjwwood, 2015-04-29:
Ok, pr2_common is in flight: https://github.com/ros/rosdistro/pull/8241

Once that's merged then you should be clear to release this @DLu, thanks for your help!

wjwwood, 2015-04-30:
@DLu I'd like to get this out today, do you have time, or should I do it?

DLu, 2015-04-30:
https://github.com/ros/rosdistro/pull/8257

closed 2015-04-30 by wjwwood