ros/urdf_tutorial #15: Wiki Changes for ROS Control Changes

DLu, 2015-06-29:
- [ ] [Minor updates to material tags and gazebo materials](http://wiki.ros.org/urdf/Tutorials/Building%20a%20Visual%20Robot%20Model%20with%20URDF%20from%20Scratch)
 - [ ] [Minor updates for dynamic parameters/gazebo dynamic tags](http://wiki.ros.org/urdf/Tutorials/Adding%20Physical%20and%20Collision%20Properties%20to%20a%20URDF%20Model)
 - [ ] [Major update with xacro cleanup](http://wiki.ros.org/urdf/Tutorials/Using%20Xacro%20to%20Clean%20Up%20a%20URDF%20File)
 - [ ] New section required: Integrating with Gazebo and ROS Control

DLu, 2015-06-29:
Can't assign @paulbovbel  for some reason.

paulbovbel, 2015-06-30:
Maybe if I'm a participant. Either way, will take a whack on Friday.

DLu, 2015-07-30:
Any progress here?

paulbovbel, 2015-08-11:
Mea culpa, not yet, will find some time asap.

closed 2017-04-29 by DLu

paulbovbel, 2017-04-29:
Eep, thank you @dlu for doing the heavy lifting.