ros/urdf_tutorial #16: missing roslaunch dependency

tkruse, 2015-07-24:
I believe since CMakeLists.txt find_packages roslaunch, it should be listed in package.xml as build_depend

tkruse, 2015-07-24:
Actually, I need to confirm, closing meanwhile.

tkruse, 2015-07-24:
Yes, without the dependency, I get
```
CMake Error at /home/kruset/work/ros/jade_sources/devel/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by "roslaunch" with
  any of the following names:

    roslaunchConfig.cmake
    roslaunch-config.cmake
```

DLu, 2015-07-24:
I can't replicate the error, but agree that it should be included. Pull request?

DLu, 2015-07-24:
Actually, it is a run dependency, so should we just remove it from the find_package command?

tkruse, 2015-07-24:
I believe it is build-required for this:
```
roslaunch_add_file_check(launch)
```

tkruse, 2015-07-24:
and the build fails accordingly:
```
CMake Error at CMakeLists.txt:7 (roslaunch_add_file_check):
  Unknown CMake command "roslaunch_add_file_check".

```

PR #17 (2015-07-24): add roslaunch to build dependencies, since it is required in cmakelist
fix #16
closed 2015-07-24 by DLu