ros/urdf_tutorial #23: Update urdf_tutorial to not depend on pr2_common repository

mikaelarguedas, 2017-04-24:
@wjwwood FYI: opening this as discussed offline

DLu, 2017-04-26:
Previous discussion was here: https://github.com/ros/urdf_tutorial/issues/13

What was the thinking offline?

mikaelarguedas, 2017-04-26:
@DLu thanks for the pointer I didn't look at the closed issues :confounded: 
The thinking offline was the following:
- It is inconvenient to depend on the entire pr2 stack just for a few resources from one of its packages
- This will allow to shrink the number of dependencies of desktop-full and curate a bit the different licences it contains (ivcon being CPL)
- It will permit to move faster in the release process of the next ros release

The details of where the resources should be put instead hasn't been discussed and I'll leave that up to @wjwwood to decide what is the best solution

DLu, 2017-04-27:
What do you say, @wjwwood? I'm thinking of copying the mesh to urdf_tutorial, but defer to higher powers.

wjwwood, 2017-04-27:
I think @mikaelarguedas has moved to get `pr2_description` released. It seems to me that referencing an external resource from another package has value in the tutorial. And referencing a "real" robot mesh makes it feel more authentic. However, copying the mesh to this repository (or just making a new gripper up), in its own package or not, is fine by me. It certainly has been annoying to have to wait on `pr2_description` each time.

commit 78320442 (2017-04-29): Remove dependency on pr2_description. Fixes #23.
closed 2017-04-29 by DLu