ros/urdf_tutorial #24: dependency on gazebo_ros

wjwwood, 2017-05-05:
While installing `ros-lunar-desktop` I was surprised to see it needing to pull in `ros-lunar-gazebo-ros` and therefore `libgazebo8`. These are being pulled in by this repository (only this one AFAICT).

From my perspective, the difference in "desktop" and "desktop-full" is that "desktop" has less dependencies, avoiding things like Stage, Gazebo, PCL, etc... This is particularly important on systems like macOS where building gazebo and PCL and stuff can be an undesired time sync.

So I'd propose that we either move these tutorials to "desktop-full" or split out the part of the tutorials that depend on `gazebo_ros` and `gazebo_ros_control`. I think those parts are useful and I would like to see them in at least "desktop-full", but I also would like to see a slightly slimmer set of dependencies for just plain "desktop".

Anyone have an opinion on what direction to take, or maybe other ideas for how to accomplish this?

@mikaelarguedas

----

One other use case I'll put forward for why this is important is that if Gazebo7 (used in Kinetic) and Gazebo8 (used in Lunar) cannot be installed at the same time (still working out if that is intended to be the case) then on a system with `ros-kinetic-desktop-full` (or more) installed cannot also install `ros-lunar-desktop` at the same time. That's really unfortunate from my perspective. I'd love to see `ros-kinetic-desktop-full` and `ros-lunar-desktop-full` side by side, but at least if desktop did not include a dependency on `gazebo_ros` it would enable the other case I described for now.

wjwwood, 2017-05-05:
Follow up on the last bit of my comments above: It appears that gazebo7 and gazebo8 conflicting is expected behavior.

DLu, 2017-05-11:
Yeah, I hadn't thought about that too much when I merged in the changes. Personally, I'm not sure why this tutorial is part of desktop full.

Even so, having gazebo as a dependency for the bulk of the tutorial is not necessarily the best. Probably the best solution would be to split this into two parts, with urdf_tutorial2 having separate dependencies. That could still be done in the same repo, I think.

mikaelarguedas, 2017-05-11:
Splitting them seems sensible to me. I think that keeping them in the same repo is ok. In the meantime, this package has been moved to desktop-full to avoid desktop to depend on the simulators: https://github.com/ros/metapackages/pull/15

DLu, 2017-07-05:
The packages have been split and released. See also urdf_sim_tutorial.

closed 2017-07-05 by DLu

clalancette, 2017-11-27:
(commenting on this issue to continue the discussion for Melodic)

[REP 150](https://github.com/ros-infrastructure/rep/pull/144) is proposing that we put both `urdf_tutorial` and `urdf_sim_tutorial` in the `desktop_full` metapackage for Melodic.  Is there any reason we would want to do something different, or does that seem acceptable?  @DLu @mikaelarguedas @wjwwood , looking for opinions here.

DLu, 2017-11-27:
To clarify, urdf_tutorial was in desktop, and now we're suggesting putting both in desktop_full?

clalancette, 2018-02-02:
@DLu Any chance you would have some time soon to do the split of the repository into `urdf_tutorial` and `urdf_sim_tutorial`?  If not, I'd be happy to help out and do it, I would just need some permissions on this repository.

DLu, 2018-02-02:
@clalancette I have full access to commit to the repository, but am not an admin so I cannot change the settings myself.

DLu, 2018-02-02:
Maybe @mikaelarguedas can give us both access?

mikaelarguedas, 2018-02-13:
@DLu @clalancette sorry I missed that poke: I created the repository at https://github.com/ros/urdf_sim_tutorial and gave rights to the robot_model team to it

clalancette, 2018-02-16:
@mikaelarguedas Excellent, thanks for that.  I've now force-pushed all of the original code from `urdf_tutorial` into the `urdf_sim_tutorial` repository.  Next up is a PR to remove the `urdf_tutorial` stuff from `urdf_sim_tutorial` repository.

mikaelarguedas, 2018-02-16:
> I've now force-pushed all of the original code from urdf_tutorial into the urdf_sim_tutorial repository. Next up is a PR to remove the urdf_tutorial stuff from urdf_sim_tutorial repository.

Was there any reason for pushing the `urdf_tutorial` package there?

clalancette, 2018-02-16:
> Was there any reason for pushing the urdf_tutorial package there?

Mostly to keep history.  We used this recipe to do all of the splitting of `robot_model`, and it worked out OK.  One slightly better thing I could would be to use `git filter-branch` to just keep the history related to `urdf_sim_tutorial`, but I wasn't sure it was worth the effort.

mikaelarguedas, 2018-02-16:
Yeah I was expecting the git-filter-branch approach, it's really not that much effort and makes the history much more relevant

clalancette, 2018-02-16:
> Yeah I was expecting the git-filter-branch approach, it's really not that much effort and makes the history much more relevant

Sure, I can do that too.  I just use `git filter-branch` so infrequently that I always have to look up all of the options :).

mikaelarguedas, 2018-02-16:
> I just use git filter-branch so infrequently that I always have to look up all of the options :).

Haha yeah I know the feeling. If it helps, this tuto is pretty good https://blogs.s-osg.org/retain-history-moving-files-git-repositories/