ros/urdf_tutorial #26: xacrodisplay.launch is missing

Kukanani, 2017-05-11:
While going through http://wiki.ros.org/urdf/Tutorials/Using%20Xacro%20to%20Clean%20Up%20a%20URDF%20File, I discovered that `xacrodisplay.launch` is now missing from the `launch` directory, causing fatal errors when trying to run the tutorial. The file is still located in `indigo`, but is missing in `default`. I can't locate the commit that removed it, but it seems to have disappeared between 1697f8c and 4b343d6.

mikaelarguedas, 2017-05-11:
The file disappeared in https://github.com/ros/urdf_tutorial/pull/14/commits/9d38b7b4da3e00bdbd17148676a86be61849293d from #14

DLu, 2017-05-17:
Closed with http://wiki.ros.org/action/diff/urdf/Tutorials/Using%20Xacro%20to%20Clean%20Up%20a%20URDF%20File?action=diff&rev1=21&rev2=22

closed 2017-05-17 by DLu