ros/urdf_tutorial #38: Problem with "RobotModel" in Rviz

ghx-0228, 2019-06-17:
I just execute the examples in your tutorial without any change. But all examples, except the 1st one, have such a problem "no transform from [] to []" as shown in the following figure.
[image]

However, when I add "RobotState", it is alright, like this.
[image]
So could you help me solve the above problem?

DLu, 2019-06-18:
The provided rviz configuration uses `/base_link` as the root frame, rather than `/map` which is the default. If you change the global fixed frame to `base_link` it will resemble the RobotState.

closed 2019-06-18 by DLu

ghx-0228, 2019-06-19:
The global fixed frame was just /base_link. The "StateModel" had many "transform" errors
[image]

DLu, 2019-06-19:
Seems like you might be out of sync with the different steps of the tutorial, as if robot_state_publisher was seeing the previous version of the URDF. It doesn't automatically reload when the parameter changes.

vec123, 2020-03-29:
Hmmm i am having a very similar issue, where i get an transform error on a seemingly correct urdf (meaning e.g. i have a first link, then a follow_up joint which specifies the child link of that fist link and the chain continues like this).  Now I expected that, when the first link is published to rviz, that then the chain would be processed, and therefore all links displayed. However this is the output:

[image]

The loaded launchfile loads the robot model like this:

`<arg name="bin" default="$(find peg_in_hole)/urdf/r2d2.urdf"/>
 <param name="r2d2/robot_description" command="$(find xacro)/xacro $(arg bin)" />
  
and the two ( as far as i know) necessary nodes like this:

  <node pkg="joint_state_publisher" type="joint_state_publisher" name="joint_state_publisher"/>
    <node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher"/>

   <node name="rviz" pkg="rviz" type="rviz" required="true" />
. 

I would expect these lines of code to load the .urdf description (which is correctly written, copied out of the tutorial) and publish the joints to /joint_states from where it goes to /tf and results in the display of the links.

I am pretty new to this so maybe i got something wrong. I can only tell that the joints never arrive at /joint_states, so i am guessing that something is missing, but i don't know what and can't seem to find anything.

`