ros/urdf_tutorial #54: fail to run the example in RoS2 foxy

yang-yuke, 2021-11-09:
Hi Dear urdf support team,

I am following this[ urdf tutorial](http://docs.ros.org/en/foxy/Tutorials/URDF/Building-a-Visual-Robot-Model-with-URDF-from-Scratch.html) on the RoS2 foxy tutorial. I have downloaded this urdf_tutorial folder to my RoS2 workspace and built it. 
[image]

Somehow, when I execute the below python script, there are some errors. 

> ros2 launch urdf_tutorial display.launch.py model:=urdf/01-myfirst.urdf

The error is as displayed in the below picture
[image]

Seems there are some inherent code errors in the files you provide, because I don't see this parameter_description module in the launch_ros package.
[image]

I haven't changed anything in the folder's files, but there are still such errors. Is it because the urdf_tutorial is still not supported in RoS2 foxy?

ilVecc, 2021-11-24:
I have the same issue! Have you managed to solve this meanwhile @yang-yuke?

ilVecc, 2021-11-24:
I have managed to solve this using
`parameters=[{'robot_description': Command(['xacro ', LaunchConfiguration('model')])}]`
and removing the unnecessary import, but now issue #55 emerged.

yang-yuke, 2021-11-27:
Hi @ilVecc,

No, I haven't resolved the issue. I think this tutorial has not yet been supported in RoS2. I don't know why they post it on RoS2's WiKi. I have stopped struggling with it. Instead, I choose to go through the tutorial in RoS1. Now I feel much better.

clalancette, 2021-11-30:
This looks to be a case where `launch_ros`, `xacro`, or `joint_state_publisher_gui` wasn't properly installed.  For what it is worth, I'm able to run this tutorial just fine on Foxy and on Rolling now that #53 has been merged.

DLu, 2021-12-06:
@yang-yuke Can you give the output of `apt-cache policy ros-foxy-launch-ros | grep Installed`

0smile, 2021-12-18:
> 

I'm facing the same problem, and here is my output:
Installed: 0.11.4-1focal.20211005.225211

yang-yuke, 2022-01-10:
> @yang-yuke Can you give the output of `apt-cache policy ros-foxy-launch-ros | grep Installed`

Hi @DLu,

The output of `apt-cache policy ros-foxy-launch-ros | grep Installed` is as below for me.
[image]

yang-yuke, 2022-01-10:
> This looks to be a case where `launch_ros`, `xacro`, or `joint_state_publisher_gui` wasn't properly installed. For what it is worth, I'm able to run this tutorial just fine on Foxy and on Rolling now that #53 has been merged.

Hi Chris,

Thank you so much!!! Your answer is short and to the point. After install Xacro with the command `sudo apt install ros-foxy-xacro`, combined with following the steps from this [webpage](https://answers.ros.org/question/328496/how-to-launch-rviz2-using-urdf-like-urdf_tutorial-of-ros1/) of this fellow(lukicdarkoo), I can see my rviz2 window will pop out with the command `ros2 launch urdf_tutorial display.launch.py model:=urdf/01-myfirst.urdf`, as displayed below.

Still, I have one thing unclear. In the below picture, I see some [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7412: messages. Is this per expected? Or does it mean something is wrong with my installation?
[image].

There is also one caveat here, just in case some rookie like me might run into it. We need to run the local_setup.bash file of this package(urdf_tutorial) before we launch its instances.

yang-yuke, 2022-01-11:
Hi Dear Mr. Chris,

I ran into an issue again. When I try to launch the urdf file, it is not working as expected. There is nothing showing up in the rviz. It is not like before. Could you please help take a look at it for me?

//Ideally, there should be a red cylinder showing up, but nothing in the rviz window. Not sure what is happening.
[image]

yang-yuke, 2022-01-17:
Try to run the example today again. Somehow, the cylinder shows up again. Just it is not so round. There are obvious corners. Looks more like a polygon rather than a round shape.
[image]

careslab, 2023-08-11:
Hi All,
I've been having the same issues as above on foxy/windows 10.  I just ran and compiled the code.  All the build and execution errors are in the attached file.
It is a great demo--it would be nice if it worked on Windows/Foxy.
[urdf_errors.pdf](https://github.com/ros/urdf_tutorial/files/12323044/urdf_errors.pdf)

Thanks!

DLu, 2023-08-14:
@careslab What version of `launch_ros` are you using?

CLOWN2255, 2024-10-14:
> 今天再次尝试运行该示例。不知何故，圆柱体再次出现。只是它不那么圆。有明显的角。看起来更像是多边形而不是圆形。 [image]

Hello, has this issue been resolved?