ros/urdf_tutorial #55: ros2 foxy problem with launching urdf_tutorial

p4lik4ri, 2021-11-22:
i am trying to launch the py file inside the launch folder but i get this error how do i fix this?
[image]

kokerf, 2021-11-23:
hello @p4lik4ri , I have the same problem! Have you already fixed it?

p4lik4ri, 2021-11-23:
hello @kokerf no i am still struggling to find a solution but nothing...but if i find one i will let you know hope you do the same

p4lik4ri, 2021-11-24:
can someone please help i am stack here i am new to ros2 and i would really appreciate some help

ilVecc, 2021-11-24:
I had this issue too, it seems like the script is passing a Path from pathlib instead of a simple string. Simply replace the default parameters to
`default_model_path = str(urdf_tutorial_path / 'urdf/01-myfirst.urdf')`
`default_rviz_config_path = str(urdf_tutorial_path / 'rviz/urdf.rviz')`
and rebuild your colcon workspace (and resource it of course :) )

In my case, this new issue emerged:
`FileNotFoundError: [Errno 2] No such file or directory: 'xacro'`
but I solved it simply by running
`sudo apt install ros-foxy-xacro`

Now everything works like a charm.

p4lik4ri, 2021-11-25:
> 
hello @ilVecc i did everything you adviced but i have the same error as previously nothing changed :(

p4lik4ri, 2021-11-25:
i did everything above but i have the same error for some reason :(

clalancette, 2021-11-30:
This is fixed by #53, so closing.

closed 2021-11-30 by clalancette