ros/urdf_tutorial #56: Release into ROS 2 Foxy, Galactic, and Rolling

clalancette, 2021-11-30:
@DLu how do you feel about releasing these packages into ROS 2 Foxy, Galactic, and Rolling?  That would make it so that the tutorials on https://docs.ros.org/en/rolling can be run without building the package.  If you think this is a good idea, I can go ahead and make the release repository in https://github.com/ros2-gbp, and can even do the release if you'd like.  Just let me know.

DLu, 2021-12-06:
I approve! I can help if needed, but its probably easiest if you make the repo and make the release(s).

clalancette, 2021-12-06:
I've created https://github.com/ros2-gbp/urdf_tutorial-release ; you should get write access to it by the end of the day.  I've also gone ahead and done releases for [Foxy](https://github.com/ros/rosdistro/pull/31389), [Galactic](https://github.com/ros/rosdistro/pull/31390), and [Rolling](https://github.com/ros/rosdistro/pull/31391).  Note that I bumped the version number for the `ros2` branch to 1.0.0; this is to give us plenty of numbers when dealing with the `ros1` branch.  With all of that done, I'm going to go ahead and close this out.

closed 2021-12-06 by clalancette