ros/urdf_tutorial #62: how install this package with ROS Humble and Gazebo Fortress

dvogureckiy99, 2023-04-03:
Hi, how install this package with ROS Humble and Gazebo Fortress ?

DLu, 2023-04-05:
This package is released with the ROS 2 binaries, so on Ubuntu `apt install ros-humble-urdf-tutorial` should work. 

If you are having other errors, please provide the exact command and the full output.

dvogureckiy99, 2023-04-05:
> This package is released with the ROS 2 binaries, so on Ubuntu `apt install ros-humble-urdf-tutorial` should work.
> 
> If you are having other errors, please provide the exact command and the full output.

E: Unable to locate package ros-humble-urdf-tutorial

DLu, 2023-04-05:
What OS?

dvogureckiy99, 2023-04-06:
> What OS?

Ubuntu 20.04

DLu, 2023-04-06:
Unfortunately binaries are not available for Humble + 20.04
https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027

Are you set on using Humble?

dvogureckiy99, 2023-04-06:
I installed Humble from source. It's not the problem.

DLu, 2023-04-10:
Then you should also install this package from source.

cccccmake, 2023-06-20:
> Then you should also install this package from source.

Hi! I've got some issue when I try building the package from source. I got the following workspace structure:

[image]

And after I run `colcon build` under the root of my workspace, I got the following outputs:

`--- stderr: urdf_tutorial                         
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.
---`

It asks for a configuration file for catkin. After I clean up the outputs of the colcon and run`catkin build`, it still asks me for catkin's config.

As for the OS, I'm running ros2 humble on arch.

Could you please offer me some suggestions regarding this issue?

DLu, 2023-08-15:
@cccccmake This appears to be an entirely separate issue, but please make sure you have the `ros2` branch checked out.

closed 2023-08-15 by DLu

AmeyaB2005, 2024-06-09:
[image]
i am getting error and unable to download urdf_tutorial

guotong1988, 2024-12-10:
ERROR:
`package 'urdf_tutorial' not found`

FIX:
`sudo apt install ros-jazzy-urdf-tutorial`

DLu, 2024-12-10:
@guotong1988 [It's there in Jazzy](https://repo.ros2.org/status_page/ros_jazzy_default.html?q=urdf_tutorial). Make sure you `sudo apt update` first.