ros/urdf_tutorial #67: Availability of Binary Packages for Windows

songyuc, 2023-10-27:
Hello,

I am working on a project using ROS 2 Iron on a Windows machine and am interested in using the `urdf_tutorial` package. I came across the [Using URDF with Robot State Publisher](https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/Using-URDF-with-Robot-State-Publisher.html) tutorial which mentions "If you installed `urdf_tutorial` binaries, this should already be the case." I am wondering if there are binary packages available for Windows and if so, how I can install them.

If the binary packages are not available for Windows, could you please provide any guidance on how to compile and use `urdf_tutorial` on Windows?

Thank you for your time and assistance.

DLu, 2023-11-07:
Apologies, I am not knowledgeable about that part of ROS 2. Perhaps @Yadunund or @clalancette has further insight.

clalancette, 2023-11-07:
Currently, we only provide binary packages on Windows for anything listed in https://github.com/ros2/ros2/blob/rolling/ros2.repos .  Since `urdf_tutorials` is not in there, there are no binary packages available for it at this time.

songyuc, 2023-11-07:
Hello,

Would it be possible to consider providing binary packages for the `urdf_tutorial` package on Windows in the future? This would greatly facilitate development for those of us working in a Windows environment.

Thank you for your time and consideration.

Best regards,
Yucheng Song

clalancette, 2023-11-07:
> Would it be possible to consider providing binary packages for the `urdf_tutorial` package on Windows in the future?

It is possible through the Robostack project: https://robostack.github.io/ .  We have long-term plans to integrate more tightly with it, but we have no definitive timeline for that.  In the meantime, you can see if RoboStack works for you.