ros/urdf_tutorial #68: Potetial Issue with launch file

zacharyyamaoka, 2023-12-13:
in https://github.com/ros/urdf_tutorial/blob/ros2/launch/display.launch.py

line 23 ->    ld.add_action(DeclareLaunchArgument(name='model', default_value=str(default_model_path),
                                        description='Path to robot urdf file relative to urdf_tutorial package'))

I belive there shouldn't be the str() there. It prevents the default value from actually working!

clalancette, 2023-12-20:
> I belive there shouldn't be the str() there. It prevents the default value from actually working!

Yep, you are absolutely correct.  Removing that `str` makes it work.  I'll submit a PR to fix it right now.  Thanks!

DLu, 2024-01-04:
Good catch, thanks!

PR #69 (2024-01-04): Fix a typo in launch that prevents it from working.
In particular, the default value for the model needs to be a substitution, not a string.  Otherwise trying to launch fails with a xacro failure.

Fixes #68
closed 2024-01-04 by DLu