ros/urdf_tutorial #73: Why isn't a cylinder circular in the first section of the URDF tutorial

CLOWN2255, 2024-10-14:
I ran the results 
[image]
The tutorial displays 
https://docs.ros.org/en/humble/Tutorials/Intermediate/URDF/Building-a-Visual-Robot-Model-with-URDF-from-Scratch.html# 
[image]

DLu, 2024-10-16:
I'm guessing something was mishandled with the `model` argument. Can you please provide the exact command you ran?

CLOWN2255, 2024-10-17:
> I'm guessing something was mishandled with the `model` argument. Can you please provide the exact command you ran?
> 
> (note that the intended syntax is in the yellowish box right above your screenshot. Running `display.launch.py` with no arguments will load a different model from the same tutorial.)

This is the command I am using
[image]

I don't know if it has anything to do with the sentence on the green line in the screenshot (Stereo is Not Supported)? I guess it might be an issue with the rviz version

DLu, 2024-10-17:
Which ROS version are you running? How is urdf tutorial installed, binary or source?